bolderflight/invensense-imu #122: assert failed: i2c_ll_set_bus_timing i2c_ll.h:109 (scl_high > 13)

lc2047, 2023-11-20:
Hi

I tried your library with i2c.ino to EPS32. The compilation is a success (no error). 
However, the system continuously reboots. 

Please share your comments with me to fix it.

`ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1416
load:0x40078000,len:14804
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c

assert failed: i2c_ll_set_bus_timing i2c_ll.h:109 (scl_high > 13)

Backtrace: 0x40082529:0x3ffb2080 0x40088335:0x3ffb20a0 0x4008d362:0x3ffb20c0 0x400dc932:0x3ffb21f0 0x400d3601:0x3ffb2210 0x400d2047:0x3ffb2240 0x400d1582:0x3ffb2260 0x400d4eb5:0x3ffb2290

ELF file SHA256: 31605a112e3abb44

Rebooting...`

lc2047, 2023-11-20:
It seems the ESP32 has some issue with I2C clock frequency. 
`Wire.setClock(400000);`
If reduce it to 100K, the code can go on.

But another issue happen.
`Error initializing communication with IMU`

flybrianfly, 2023-11-21:
That's just the code's way of telling you that you aren't communicating with the sensor. The ESP32 core software used to have a problem with endTransmission(false) - I'm not sure if they've fixed it yet on their end, but you may need to update this library to use endTransmission() instead of endTransmission(false). Unfortunately, the sensor data sheet tells us to use that as part of the communication protocol, but it's up to the chip manufacturers to implement it correctly in their core software - this is just a work around that I know has helped people in the past.

It could also be an issue with your hardware setup, so I would run an I2C scanner example to check that you can see the sensor.

lc2047, 2023-11-24:
Hi @flybrianfly 

Thanks for your support. I tried many codes and modified code for pass-through. ALL are not successful.
I remember somebody had a similar problem and fixed the issue with a sample. It became true in my case.
Why do a lot of problems at AK8963 only? I can't believe the reliability of TDK to such a level.

ruairc-p, 2023-11-24:
Yeah Ic2047
You would think that someone selling a product would have an interest in making sure people could use it. Unless they had some anti-competitive, anti free-market reason for doing so. 
Crazy World.

flybrianfly, 2023-11-24:
@lc2047, I would try with a different microcontroller if you have one available - I don't have any issues with the Teensy and STM32 microcontrollers. I do think the I2C library is not as well implemented in the ESP32 core. Alternatively, you could try using SPI instead of I2C.

Finally, check that you got the MPU-9250 from a reputable source, there's a lot of fake ones out there.