bolderflight/invensense-imu #123: Please fix 1.0.2 release

okalachev, 2024-01-03:
Hi! Thanks for the library, for now I use the 1.x branch of it in my project of open source [ESP32 based quadcopter](https://github.com/okalachev/flix). I'm not planning to switch to the newer version yet.

The problem is that the most recent 1.x release – 1.0.2 has a problem with [incorrect version](https://github.com/bolderflight/invensense-imu/blob/v1.0.2/library.properties#L2) value in `library.properties`, thus it's not shown up in Library Manager in Arduino IDE. Here is this error in the log: https://downloads.arduino.cc/libraries/logs/github.com/bolderflight/MPU9250.

While this version has a critical [bug fix](https://github.com/bolderflight/invensense-imu/commit/7031a2056efc89b7d9588f0c0eb02cd6947d6b57) in accelerometer calibration.

Can you please fix version value for this tag? Or create a new tag, like 1.0.4 (1.0.3 is occupied with the newer API version). And then trigger the library-registry to get everything updated.

flybrianfly, 2024-01-04:
v1.0.3 looks like it should work, is that an option? Alternatively, you could fork v1.0.2, fix the tag, and have the Arduino library manager use that.

I'm a little hesitant to support older versions, even if it's a minor change. The only way I can figure out how to do this would be to have the v1.0 version on its own branch with the tag fix applied, I don't see an easy way for that to reside on the main branch.

okalachev, 2024-01-04:
v1.0.3 looks like kind of a mistake, it has totally different API in comparison with 1.0.2, I mean it has the 6.x version API, so it doesn't work for me.

I made a fork, but until I register this fork in Arduino registry, it doesn't appear in Library Manager of Arduino IDE. And registering a fork, with literally ONE byte of difference from the official version looks kind of weird solution.

The easy solution would be just moving the tag `v1.0.2` from commit 57dccc77caa2051e25a51ed2febb764aee5b6af2 to the next commit 78c33712a73643f2364ab11290e1a77c98a1e005 where you exactly fixed that issue with version number in library.properties.

okalachev, 2024-01-04:
I even think that in the past that tag was moved on the 78c33712a73643f2364ab11290e1a77c98a1e005 commit, but was accidentally moved back by someone who inaccurately pushed the old tag to the repo?

flybrianfly, 2024-01-04:
> v1.0.3 looks like kind of a mistake, it has totally different API in comparison with 1.0.2, I mean it has the 6.x version API, so it doesn't work for me.
> 
> I made a fork, but until I register this fork in Arduino registry, it doesn't appear in Library Manager of Arduino IDE. And registering a fork, with literally ONE byte of difference from the official version looks kind of weird solution.
> 
> The easy solution would be just moving the tag `v1.0.2` from commit [57dccc7](https://github.com/bolderflight/invensense-imu/commit/57dccc77caa2051e25a51ed2febb764aee5b6af2) to the next commit [78c3371](https://github.com/bolderflight/invensense-imu/commit/78c33712a73643f2364ab11290e1a77c98a1e005) where you exactly fixed that issue with version number in `library.properties`. It would totally fix everything.

Thanks for that hint. I moved the v1.0.2 tag to the 78C3371 commit.

closed 2024-01-05 by okalachev