bolderflight/invensense-imu #124: Compilation Error on Arduino-Esp32

enoch-prince, 2024-05-08:
Compilation of the I2C example for the ESP-WROOM-32 Dev Board using the Arduino 2.3.2 IDE, failed with exit status 1 and the message:
> cc1plus.exe: some warnings being treated as errors

Details of the error log is below:

```C++
c:\Users\LENOVO T460S\Documents\Arduino\libraries\Bolder_Flight_Systems_Unit_Conversions\src\convangacc.cpp: In function 'float bfs::convangacc(float, bfs::AngAccUnit, bfs::AngAccUnit)':
c:\Users\LENOVO T460S\Documents\Arduino\libraries\Bolder_Flight_Systems_Unit_Conversions\src\convangacc.cpp:59:9: error: 'out_val' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   float out_val;
         ^~~~~~~
c:\Users\LENOVO T460S\Documents\Arduino\libraries\Bolder_Flight_Systems_Unit_Conversions\src\convangacc.cpp:70:24: error: 'in_val' may be used uninitialized in this function [-Werror=maybe-uninitialized]
       out_val = in_val / BFS_2PI_FLOAT * 60.0f;
                 ~~~~~~~^~~~~~~~~~~~~~~
cc1plus.exe: some warnings being treated as errors
```

enoch-prince, 2024-05-08:
Versions used
- Units Conversion: V5.0.0
- Eigen: V3.0.2
- Invensense-IMU: V5.6.0

Yet still get the same compile errors as described above.

Furthermore, the following files in the Units library also produce the same errors:
```C++
convdensity.cpp
convangacc.cpp
convacc.cpp
convang.cpp
convangvel.cpp
```

flybrianfly, 2024-05-08:
The current version of the InvenSense IMU library (v6.0.3) does not need Eigen or Units.

closed 2024-05-08 by enoch-prince