bolderflight/invensense-imu #125: Error Verifying The Code In Arduino Cloud IDE

dude363635, 2024-11-02:
I get an error about dependencies and how there are multiple of them when i try to verify the code in Arduino Cloud IDE. Pic below:
[image]
Could you please help me? I'm new to Arduino and i need this for a project.

flybrianfly, 2024-11-02:
You would need to copy and paste the entire error output, I'm not seeing any errors here.

flybrianfly, 2024-11-14:
The newer versions of the library include cstddef, but use include guards so it shouldn't be used in an Arduino build:

```C++
#include "mpu9250.h"  // NOLINT
#if defined(ARDUINO)
#include <Arduino.h>
#include "Wire.h"
#include "SPI.h"
#else
#include <cstddef>
#include <cstdint>
#include <algorithm>
#include "core/core.h"
#endif
```

Maybe the Arduino Cloud IDE isn't including `ARDUINO`?

rizacelik, 2024-11-14:
When we install the library with Arduino IDE, the coding in the mpu9250.h file is as follows. When I look at the repository, the mpu9250.h coding is in the format you provided. Where does Arduino IDE download your library codes from? What is the reason for this difference?

```

#if defined(ARDUINO)
#include <Arduino.h>
#include "Wire.h"
#include "SPI.h"
#else
#include "core/core.h"
#endif
#include <cstddef>
#include <cstdint>
#include "eigen.h"  // NOLINT
#include "Eigen/Dense"

```